manna-harbour/miryoku #3: [Feature Request] Support on ZMK

harshitgoel96, 2021-04-21:
Can this layout be made available on ZMK?

## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Percentnineteen, 2021-06-17:
I'm setting up a Corne on ZMK and was wondering if I could use your dev version as a starting point for my keycap. I was also wondering if you planned to work in bluetooth controls for ZMK given its focus on wireless functionality. Thanks.

manna-harbour, 2021-06-18:
> I'm setting up a Corne on ZMK and was wondering if I could use your dev version as a starting point for my keycap. I was also wondering if you planned to work in bluetooth controls for ZMK given its focus on wireless functionality. Thanks.

Sure.  Please contact me privately to arrange this.  https://github.com/manna-harbour/manna-harbour/tree/main/contact

Bluetooth controls are on the media layer for the time being.

closed 2021-11-30 by manna-harbour