manna-harbour/miryoku #5: Feature Request (180 degrees keyboard rotation)

JhorAVi, 2021-07-23:
Most grid keyboards like planck and preonic have USB cables located on top which takes a lot of space between laptop and keyboard resulting to screen pushed farther away.  Rotating the keyboard 180 degrees removes the wire obstacle allowing the keyboard to touch the laptop for more stable support. I have successfully rotated the keyboard by editing the layout map and typing experience is much better actually. Can a rotation define be added to make editing easier?

manna-harbour, 2021-07-26:
That's a novel use for subset mapping!  I guess a right angled cable would almost but not quite do the same thing.  I can add this but it needs to be added to the subset mapping, so per keyboard or layout.  Where would you use it now?  `LAYOUT_ortho_4x12`?

JhorAVi, 2021-07-26:
The right angled cable is just a partial solution because wire still exists which doesn't allow me to push the keyboard against the laptop for more stable typing. 
The keyboard I'm using now is 4x10  pabile/p40_ortho
Thank you.