manna-harbour/miryoku #33: Mods Customisations

manna-harbour, 2021-11-25:
Support various mods customisations. E.g.

- [ ] Ctrl / Gui swap for Mac for implementations lacking that feature
- [ ] Reordering mods for user preference
- [ ] Disabling mod-tap to prevent accidental mods (use mods on sub layers)
- [ ] Enabling one shot mods on sub layers
- [ ] Removing AltGr when not needed
- [ ] Per-key tapping term
- [ ] Per-key mod tap behaviour

manna-harbour, 2022-02-21:
Ctrl / Gui swap for Mac might not be necessary as per keyboard mod remapping is supported in macOS: https://support.apple.com/en-az/guide/mac-help/mchlp1011/mac

bijou-code, 2023-05-01:
I'm new to the community (this weekend), and I don't know if there is a better place to put this (like in discussions), but I wanted to share my solution for editing the home row mod order after being redirected from [this reddit post](https://www.reddit.com/r/ErgoMechKeyboards/comments/ugyf0h/miryoku_is_there_a_good_way_to_change_the_home/) and one of the discussions in [this repo](https://github.com/manna-harbour/miryoku/discussions/34). This modification was done in my QMK installation since I prefer GASC to GACS (using colemak-dh), which isn't generally/easily OS configurable.

Notes:
- I performed these changes against the babel files so that all keyboards I create
- I have not tried these modifications in ZMK, but it looks like the same commands _should_ work if the `sed` strings are tweaked.
- These commands rely on the default ordering of the home row mods, but can easily be modified to account for existing changes in the `*layer_alternatives.h` file.
- These commands are not idempotent and may result in a different/ random state if executed multiple times without a reset.
- These commands appear to be repeatable, and should be able to be re-applied after accepting a force-update from upstream.

``` bash
# get to the qmk installation root
cd && qmk cd

# set each of the mods to a temporary value and repopulate with the mods in the preferred order
sed -e 's/LGUI/TMP0/g' -e 's/LALT/TMP1/g' -e 's/LCTL/TMP2/g' -e 's/LSFT/TMP3/g' \
-e 's/TMP0/LGUI/g' \
-e 's/TMP1/LALT/g' \
-e 's/TMP2/LSFT/g' \
-e 's/TMP3/LCTL/g' \
users/manna-harbour_miryoku/miryoku_babel/miryoku_layer_alternatives.h > miryoku_layer_alternatives_temp.h

# verify the results are what you expected. If you're in a clean git state, you can copy first and use git diff instead
diff users/manna-harbour_miryoku/miryoku_babel/miryoku_layer_alternatives.h  miryoku_layer_alternatives_temp.h | less

# move the results over top of the original
mv miryoku_layer_alternatives_temp.h users/manna-harbour_miryoku/miryoku_babel/miryoku_layer_alternatives.h

# if something goes wrong, revert to "master" version. You can use a specific commit or "HEAD" instead
git checkout master -- users/manna-harbour_miryoku/miryoku_babel/miryoku_layer_alternatives.h
```

postylem, 2023-12-08:
> Ctrl / Gui swap for Mac might not be necessary as per keyboard mod remapping is supported in macOS: https://support.apple.com/en-az/guide/mac-help/mchlp1011/mac
>
> An additional pre-swapped Mac clipboard option might be necessary.

I am using macOS, and I swapped <kbd>control</kbd> and <kbd>command</kbd> in macOS for my keyboard, as you suggested.  But then I found that the clipboard commands no longer worked (because `U_UNDO` sent <kbd>control</kbd>+<kbd>z</kbd> rather than <kbd>command</kbd>+<kbd>z</kbd>, etc.).  So I added the following, and now it works:

- in `manna-harbour_miryoku.h` I defined a new clipboard option "MIRYOKU_CLIPBOARD_MACSWAP"

    [image]

- in `custom_rules.mk` I specified to use it:

    ```mk
    MIRYOKU_CLIPBOARD=MACSWAP
    ```

It took me a little while to figure out how to do that, so I'm posting it here in case it's helpful for others.  Perhaps this clipboard option could be included by default?  I would imagine other mac users want to do similarly.

manna-harbour, 2024-06-17:
Consider:

* Supporting right mods on bottom row: https://github.com/manna-harbour/miryoku/discussions/262#discussioncomment-9504589.
* Option to remove HRM and add bottom row mods.
* Moving layer lock to inner index column.