manna-harbour/miryoku #36: Implement Auto Shift via Tap Hold

manna-harbour, 2021-11-25:
Auto shift can be implemented via tap hold in ZMK and KMonad where there is no native feature,

manna-harbour, 2021-11-25:
For ZMK it can be implemented with a macro and a custom hold tap behaviour, see https://github.com/serebrov/zmk-config/blob/master/config/microdox.keymap#L23-L36.

For KMonad this was already in the proof of concept implementation.

manna-harbour, 2022-03-07:
> what is autoshift used for in this layout? looking at the settings it is disable for symbols and letters.

It's used as an alternative to shift on the Num layer. It's disabled for alphas as it's not compatible with mod tap without Retro Shift, so it would only work on some alphas. See:
- https://github.com/manna-harbour/miryoku_qmk/blob/f5c215eff82c916cc040ea884f2ec37d2a3336fc/users/manna-harbour_miryoku/config.h#L19
- https://github.com/manna-harbour/miryoku/blob/e2fec0203a3750d201bb46fc4308251c64b8db10/docs/reference/readme.org?plain=1#L34

Vasilis, 2022-03-27:
Hi, everyone! I recently enabled auto-shift on my kb. That works fine except for the copy - paste commands where instead of LGUI+C, what is sent is LGUI+SHIFT+C if I accidentally hold it for longer. 

Is it possible to always prevent sending shift (or prevent the hol-tap behavior altogether) when LGUI is pressed before?

manna-harbour, 2022-03-28:
@Vasilis I won't be enabling it for alphas as it won't work where there is mod tap.  Do you get accidental shifts in normal typing?  You could try combining with mod morph, create a new tap hold behaviour, increase the timeout, use quick snappy taps, or use the clipboard keys on the Nav layer.

Vasilis, 2022-03-28:
Thanks for following up. I am tracking the issue [here](https://github.com/zmkfirmware/zmk/issues/1190#issuecomment-1080208533).

jroes, 2025-01-08:
Auto shift is not enabled by default, even on numbers. I'm not sure if that was a recent change or not, but as of the latest miryoku for zmk that I could download, I was not able to activate it. That said, I've never used auto shift before, so I'm not entirely sure. But with my keyboard flashed to miryoku zmk, when I press and hold on a number from the Num layer, it doesn't convert to its shifted form, it just repeats the key.

Is this expected?

manna-harbour, 2025-02-12:
@jroes Yes, that's what this issue is for: ZMK doesn't have native auto shift, so it needs to be implemented manually.