manna-harbour/miryoku #51: Add Caps Word

manna-harbour, 2021-11-30:


BorjaLL, 2022-02-25:
any updates on this?

manna-harbour, 2022-02-26:
> any updates on this?

I expect to add it in the next few weeks. I've added some todos above so you can keep track of progress.

BorjaLL, 2022-03-02:
I managed to add it here. https://github.com/BorjaLL/qmk_files/tree/main/crkbd/keymaps/borjall
maybe that's helpfull

manna-harbour, 2022-03-07:
> I managed to add it here. https://github.com/BorjaLL/qmk_files/tree/main/crkbd/keymaps/borjall maybe that's helpfull

Thanks, there are a few implementations for QMK, but https://getreuer.info/posts/keyboards/caps-word/ is definitely one of the candidates.

getreuer, 2022-05-19:
There is progress on that implementation checklist!

* https://github.com/qmk/qmk_firmware/pull/16588 is now merged into QMK's develop branch. This should merge into master on 2022-05-28.
* [ZMK has a Caps Word implementation](https://zmk.dev/docs/behaviors/caps-word) already. I believe it was released early this year.
* (I'm not aware of a Caps Word implementation for KMonad. :-/)

manna-harbour, 2022-05-20:
@getreuer Thanks for the QMK PR and the update! I'll try to have it ready for the 28th! For ZMK, it's been added in the middle of a bunch of implementation changes and new features and they'll be pushed all together in a few days.

manna-harbour, 2022-05-23:
Caps Word is now available in Miryoku ZMK, see the Miryoku ZMK readme for details.

azzamsa, 2022-06-27:
Sorry to bump the issue without any meaningful help. 

Any update for caps word QMK?

azzamsa, 2022-06-28:
I got it working with VI navigation https://github.com/azzamsa/miryoku_qmk/commit/4324456c77dbe2472e97aa7527e5c0895c795068, but failed with caps word https://github.com/azzamsa/miryoku_qmk/commit/caf66d38e870b6eddddeb8843bcefb61dc41036b 

Am I missing something?

manna-harbour, 2022-06-29:
@azzamsa For caps word, first try only the "To try it now" paragraph above. You'll need to do the 3 things mentioned there exactly. See the example workflow options linked above.

If you don't use the workflow you'll still need the same 2 file changes but will need to rebase manually instead. If you want to do that, see https://github.com/manna-harbour/miryoku_qmk/blob/e9dfa7d007e9655ebe6ef7c22bc6112eb228782a/.github/workflows/main.yml#L60-L61 for how to drop the `[local]` commits first in order to avoid merge conflicts.

If that works, you can try the "To replace" paragraph. For that, what you have here https://github.com/azzamsa/miryoku_qmk/blob/caf66d38e870b6eddddeb8843bcefb61dc41036b/users/manna-harbour_miryoku/custom_config.h looks good except the macro name needs to be changed to `MIRYOKU_LAYER_NAV` here https://github.com/azzamsa/miryoku_qmk/blob/caf66d38e870b6eddddeb8843bcefb61dc41036b/users/manna-harbour_miryoku/custom_config.h#L8.

azzamsa, 2022-06-30:
> @azzamsa For caps word, first try only the "To try it now" paragraph above. You'll need to do the 3 things mentioned there exactly. See the example workflow options linked above.

Yes, I have did those three things. The hardest thing is the merge conflict. Many of the github workflows deleted upstream. So I just use magit featur to just use "their" state. 

Now, I just need to add this change:

```diff
 #pragma once
 
-#define DOUBLE_TAP_SHIFT_TURNS_ON_CAPS_WORD \
+#define DOUBLE_TAP_SHIFT_TURNS_ON_CAPS_WORD
+
+#define MIRYOKU_LAYER_NAV \
 RESET,             DF(U_TAP),         DF(U_EXTRA),       DF(U_BASE),        U_NA,              U_RDO,             U_PST,             U_CPY,             U_CUT,             U_UND,             \
 KC_LGUI,           KC_LALT,           KC_LCTL,           KC_LSFT,           U_NA,              KC_LEFT,           KC_DOWN,           KC_UP,             KC_RGHT,           CAPS_WORD,         \
 U_NA,              KC_ALGR,           DF(U_NUM),         DF(U_NAV),         U_NA,              KC_HOME,           KC_PGDN,           KC_PGUP,           KC_END,            KC_INS,            \
```

manna-harbour, 2022-06-30:
The workflows handle the merge automatically. If using magit you could interactive rebase and drop all `[local]` commits first, then there won't be any conflicts when merging with upstream master. Those commits only apply to the Miryoku QMK fork.

With `DOUBLE_TAP_SHIFT_TURNS_ON_CAPS_WORD` you can double tap shift (shift on sublayers are basic keycodes and can be tapped) to activate caps word. With the custom Nav you're replacing caps lock with caps word.

They are both just if you want to try out caps word now, with the minimum of changes. When I add it it'll be with the shift function for caps lock, but that will require using key overrides https://docs.qmk.fm/#/feature_key_overrides.

azzamsa, 2022-06-30:
Yes, I have tried it. It works well. :)

Now, I reverted my branch to get usual caps lock. Will wait the "official" release.

getreuer, 2022-09-30:
Another update: Caps Word is now exposed in Oryx ([announcement](https://blog.zsa.io/2206-caps-word/)). So that addresses the implementation checkbox for Miryoku Oryx.

manna-harbour, 2022-09-30:
@getreuer Excellent, thanks for the update! I love ticking those checkboxes! I'll try to add it to Miryoku QMK and Oryx next week (and tick more boxes! 😂).

manna-harbour, 2022-11-05:
Caps Word in now available in Miryoku QMK in a temporary banch: https://github.com/manna-harbour/miryoku_qmk/tree/capsword-tmp. It'll be added to the `miryoku` branch in a couple of days. Thanks everyone for your patience!

manna-harbour, 2022-11-07:
Caps Word in now used in place of Caps Lock in Miryoku QMK and Miryoku ZMK. Combine with Shift for Caps Lock.

KMonad doesn't support Caps Word, and Oryx doesn't support key overrides (for combining with Shift for Caps Lock), so those tasks have been moved to https://github.com/manna-harbour/miryoku/issues/90 and this issue will be closed.

closed 2022-11-07 by manna-harbour