manna-harbour/miryoku #52: Add Layer Lock

manna-harbour, 2021-11-30:
See #15.

manna-harbour, 2022-05-13:
Current layer lock is not supported in ZMK, see https://github.com/zmkfirmware/zmk/issues/1299. Opposite layer lock can still be used.

manna-harbour, 2022-05-23:
Layer lock for current and opposite layers has been added. See https://github.com/manna-harbour/miryoku/tree/master/docs/reference#additional-features.

closed 2022-05-23 by manna-harbour

manna-harbour, 2022-11-07:
Layer lock and other Additional Features now require a double tap to activate to prevent accidental activation.

maxesser1776, 2025-03-09:
@manna-harbour can you point me to the source code for the layer lock activation I'm trying to implement this in my own keymap thank you!!!

manna-harbour, 2025-03-09:
@maxesser1776 Double tap to activate is a tap dance. The Miryoku ZMK Implementation is here: https://github.com/manna-harbour/miryoku_zmk/blob/master/miryoku/miryoku_double_tap_guard.dtsi, https://github.com/manna-harbour/miryoku_zmk/blob/master/miryoku/miryoku_double_tap_guard.h.

That implementation is very indirect. For your own keymap refer to the docs and just use the behavior directly without any `cpp` macros: https://zmk.dev/docs/keymaps/behaviors/tap-dance.

For the layer lock itself see https://github.com/manna-harbour/miryoku_zmk?tab=readme-ov-file#no-current-layer-lock.