manna-harbour/miryoku #56: Add option to emulate inner index and pinkie column keys with combos

manna-harbour, 2021-12-14:
Add an option to emulate the inner index keys with combos of index and middle, and the pinkie column keys with combos of middle and ring. the inner index.  This would be an alternative to stretching, and

manna-harbour, 2021-12-14:
- Would support anything from hard to reach top row pinkie column keys to 3x3+2.
- Holding simultaneous mod combinations alt+ctrl, or ctrl+shift would not be possible as they would trigger a combo, but they could be held sequentially with a delay longer than combo term.
- If using only 3 columns it'd be better to use the inverted T nav option to reduce the need for combos.
- Generate combos from Miryoku Babel.
- Adding direct support for combo emulation of other keys to QMK and ZMK would simplify the keymap.

manna-harbour, 2021-12-14:
- [ ] Add hand written option to Miryoku ZMK for testing.  Default alternative layout options only.

distracteddev, 2022-11-20:
Is there an option to enable this feature in ZMK already or is it still a wip?

manna-harbour, 2022-11-20:
@distracteddev In general see https://github.com/manna-harbour/miryoku/issues/56#issuecomment-993376643 above for general status and a link to the docs. 

There are a few other undocumented things to note: 

It relies on support being added to the mapping file. Check the file for your keyboard and if there's nothing there support will be added by request.

Thumb combos should support all build options, but top and bottom row combos are only included for default options at the moment. QWERTY will be added soon via https://github.com/manna-harbour/miryoku_zmk/pull/8. Support for other build options will be added by request.

Some boards have combos enabled by default but they can be enabled with the appropriate `#define` in `custom_config` on other boards that have support in the mapping file. https://github.com/manna-harbour/miryoku_zmk/blob/master/miryoku/mapping/34/ferris.h has thumb combos defined and enabled. https://github.com/manna-harbour/miryoku_zmk/blob/master/miryoku/mapping/30/hummingbird.h has all combos defined, and bottom row and thumb combos enabled.

Boards with combos enabled by default should already have max combos set. Otherwise you need to set it in kconfig.

manna-harbour, 2022-11-20:
@distracteddev https://github.com/manna-harbour/miryoku_zmk/pull/8 also includes some of the currently missing docs: https://github.com/manna-harbour/miryoku_zmk/blob/d40b3e6f30d5f4652aec2583483af93dd8b19175/readme.org#key-emulation-combos.