manna-harbour/miryoku #72: Layout Diagrams

manna-harbour, 2022-02-04:
- [ ] Auto generate from Miryoku Babel
- [ ] Suitable for printing

manna-harbour, 2022-02-04:
To manually modify the current diagrams:

1. Find the JSON source under https://github.com/manna-harbour/miryoku/tree/master/data/layers
2. Download the JSON source, e.g https://raw.githubusercontent.com/manna-harbour/miryoku/master/data/layers/miryoku-kle-reference.json
3. Open http://www.keyboard-layout-editor.com/
4. Select the `</> Raw data` tab
5. Press the `Upload JSON` button
6. Upload the JSON source
6. Edit (See `Help & keyboard shortcuts`)
7. Press the `Download` button
7. Select `Download PNG` and `Download JSON`

manna-harbour, 2022-02-04:
keyboard-layout-editor.com does not support high resolution downloads.  A work around is to zoom and screenshot.

amerikiwi93, 2022-02-09:
appreciate the info. looks like http://www.keyboard-layout-editor.com/ is broken. when I upload the json file nothing happens.

jwhendy, 2022-03-22:
Since this issue already exists, I wanted to add this minor fix. I'm brand new to this layout, so maybe I'm not interpreting it right.

Base layer image from readme:
[image]

The explanation just below the image:
> The primary, secondary, and tertiary thumb keys are `backspace`, `enter`, and `delete` on the right, and `space`, `tab`, and `escape` on the left, respectively.

Compared to the text description, the layout looks like: `enter`, `backspace`, `delete`  to me.

manna-harbour, 2022-03-25:
@jwhendy Thanks for raising the issue!  The problem is that the Reference Manual doesn't explicitly state which are the primary, secondary, and tertiary thumb keys (although it is implied later on when describing individual layers).  I'll be adding a User Guide that will introduce the fundamentals without any assumptions on prior knowledge, and will use a diagram such as this to illustrate the thumb keys:

[image]

If you'd like like to discuss this further, please create a new discussion (see https://github.com/manna-harbour/miryoku/discussions/8).

jwhendy, 2022-03-25:
Whoops! Thanks for clarifying and this makes complete sense now :) It struck me as a good bit of mental fun, reminding me that even after checking one's mental model, it can still be wrong in unexpected (and yet such simple!) ways :)

My first ergo board shows up tomorrow and I look forward to revisiting the layout (perhaps on the way to a smaller board), as I have run across this many times in my researching. Thanks again!

manna-harbour, 2022-06-14:
This is currently in development and an SVG diagram showing all layers individually can now be generated for any combination of alternative layout options.

If you'd like to use this now [contact me directly](https://github.com/manna-harbour/manna-harbour/blob/main/contact) and specify the alternative layout options and I'll send you the corresponding SVG.

VitMain, 2023-12-16:
I tried to contact you privately in mail, and discord pm, but no response.

evanjrowley, 2024-07-21:
Any update on this?

It would be a big help for me to have a reference diagram for these miryoku_zmk options (taken from the GH actions process-inputs:

```
  set-output "board" "nice_nano_v2"
  set-output "shield" "cradio_left,cradio_right"
  set-output "alphas" "Dvorak"
```

manna-harbour, 2024-07-22:
@evanjrowley There's no release yet, but if you contact me via email or discord I can send you the PDF: https://github.com/manna-harbour/miryoku/issues/72#issuecomment-1154588220.