manna-harbour/miryoku #90: Miscellaneous Small Issues

manna-harbour, 2022-06-03:
### Miryoku

- [ ] Add note to [Base](https://github.com/manna-harbour/miryoku/tree/master/docs/reference#base) to use shift with `,`, `.`, and `/` to get `<`, `>`, and `?` 
- [ ] For alternative layout and mapping options, make default explicit and `#error` if no match

### Miryoku QMK

- [x] Move `layers` enum to `manna-harbour_miryoku.h`
- [ ] Add `#define LAYER_STATE_16BIT`
- [ ] Add note to update submodules when changing branches, or merge into upstream master instead
- [ ] Add note about merging `miryoku` branch with upstream QMK master:  cherry pick instead, or drop `[local]` commits first
- [ ] Add note on erasing EEPROM if non-0 default layer set in previous keymap (`set_single_persistent_default_layer()`)
- [ ] Turn off all unused features that might be enabled by default at the keyboard level to reduce firmware size
- [ ] Investigate via / vial
- [ ] Add docs on getting started with a new keyboard
- [x] Change workflow to append to custom config and custom rules
- [ ] Document docker builds
- [ ] Disable auto shift when on Tap layer, from #18 

### Miryoku ZMK

- [ ] Add outboard docs
- [ ] Add multiline Kconfig workflow example

### Miryoku KMonad

- [ ] Add support to substitute `defsrc` in `custom_config.h`

### Miryoku Babel

### Miryoku Oryx

- [ ] Add Button layer
- [ ] Add Extra layer
- [ ] Add Tap layer
- [ ] Add layer lock
- [ ] Change mouse button order

fearoffish, 2022-09-29:
I'm not sure if this helps, but I set up some of what I could in VIAL on my Cantor (ignoring the 6 extra keys on the sides, as I have the 42 key variant). It's a good starting point, maybe, so I thought I'd upload it for you/everyone. I had to add `.txt` to the filename to get it to upload here, but it's a Vial file.
[miryoku.vil.txt](https://github.com/manna-harbour/miryoku/files/9679483/miryoku.vil.txt)

javjar, 2023-05-17:
Hi! Any updates on the Kaleidoscope support plan?

mosnamarco, 2024-05-29:
yo! any updates with the vial/via investigation?

manna-harbour, 2024-05-30:
@mosnamarco @javjar Please create a discussion in Q&A for any questions.
@fearoffish Please create a discussion in Show and Tell to share your config.