manna-harbour/miryoku #94: Support additional customisation in custom_config.h

manna-harbour, 2022-06-09:
- [ ] change tapping term
- [ ] mods customisation https://github.com/manna-harbour/miryoku/issues/33
- [ ] custom clipboard? including https://github.com/manna-harbour/miryoku/issues/33#issuecomment-1046352863
- [ ] disable reset, default layer, layer lock? https://github.com/manna-harbour/miryoku/discussions/93
- [ ] add custom combos
- [ ] disable auto shift, including https://github.com/manna-harbour/miryoku/issues/36
- [ ] https://github.com/manna-harbour/miryoku/issues/35

manna-harbour, 2022-06-09:
@agisga Continuing from https://github.com/manna-harbour/miryoku/discussions/85#discussioncomment-2834628.
There's no extra support for custom combos in Miryoku QMK at the moment, but they can be added in `manna-harbour_miryoku.c` as you've been doing without too much risk of merge conflicts. I may add `custom_keymap.c` or similar for custom userspace code such as this. But in Miryoku ZMK I've defined macros for the key emulation combos, so something like that in Miryoku ZMK and Miryoku QMK for user combos might help. Or to keep it in `custom_config.h` only it could be done via X-macros in the same way as the layer list.

nodupe, 2023-11-12:
Hi, is there still a plan for adding custom_keymap.c? That would be really cool.