manna-harbour/miryoku #111: NOREVERSEANGLE option causes kmonad parse error

orand, 2022-07-26:
Normal GitHub Actions workflow builds are working fine for me.  But if I try to use the NOREVERSEANGLE option, I receive a parse error from kmonad when I try to run it:

```
kmonad: Parse error at 15:3:
   |
15 |   tap q w e r i o p [ ]
   |   ^
unexpected 't'
expecting ')'
```

For context, here are more of the lines around this:

```
(defsrc
  grv 1 2 3 4 8 9 0 - =
  tap q w e r i o p [ ]
  caps a s d f k l ; ' ent
                z x c , . /
)
```

I'm not up to speed on the syntax, but it doesn't like the `t` in `tap`.

If you want to repro it, these are the parameters for the workflow I'm using:

```
MIRYOKU_CLIPBOARD=MAC
MIRYOKU_MAPPING=NOREVERSEANGLE
MIRYOKU_KMONAD_OS=MAC
```

manna-harbour, 2022-07-26:
Thanks for the report!

That looks like a typo, and `tap` should be changed to `tab` here: https://github.com/manna-harbour/miryoku_kmonad/blob/f5dbfe74f0eebe05f9655c5415fdf17ab3fb057f/src/miryoku_kmonad.kbd.cpp#L27

orand, 2022-07-26:
Thanks, that was it!  It works perfectly now.  Pull request submitted.

closed 2022-07-28 by manna-harbour