manna-harbour/miryoku #144: macOS and difference between miryoku QMK and ZMK

xbladesub, 2022-09-23:
I was using QMK for a while and after I built my 2nd board with nice-nanos I realized that Miryoku ZMK doesn't seems to works the same as miryouku qmk - for example I can't move my mouse pointer with zmk (there is no such issues  with qmk). Also karabiner keybindings doesn't work with miryoku zmk (everything is ok with qmk). 

All settings just default (only vi mode). Am I doing something wrong? I think miryouku zmk/qmk should have same behavior with same workflow settings?

xbladesub, 2022-09-23:
Cannot use mouse-pr branch because of merge conflict 

[image]

manna-harbour, 2022-09-24:
Issues are not being used for support. Please use the Discussions as described at https://github.com/manna-harbour/miryoku#discussions-and-support.

closed 2023-02-26 by manna-harbour