manna-harbour/miryoku #207: Merge conflict when merging with QMK :(

stratosgear, 2023-04-07:
Hi,

I am new to miryoku and I am trying to do a very basic vanilla build for mty new Kyria keyboard.

If I read the instructions correctly I would have to merge QMK and miryoku_qmk  branches together with:

```
git clone git@github.com:qmk/qmk_firmware.git
cd qmk_firmware
git remote add miryoku_qmk git@github.com:manna-harbour/miryoku_qmk.git
git fetch miryoku_qmk
git checkout --track miryoku_qmk/miryoku
make git-submodule
git checkout -b miryoku-merge-master
git revert --no-edit `git log --grep='^\[miryoku-github\]' --pretty='format:%H' | tr '\n' ' '`
git fetch origin  # Maybe not even needed, as I just did a checkout 
git merge origin/master
```

But when I try it on a brand new directory, I get the following error after the last command:

```
Auto-merging users/manna-harbour_miryoku/manna-harbour_miryoku.c
CONFLICT (content): Merge conflict in users/manna-harbour_miryoku/manna-harbour_miryoku.c
Automatic merge failed; fix conflicts and then commit the result.
```

Checking the differences I see:

```
diff --cc users/manna-harbour_miryoku/manna-harbour_miryoku.c
index 5de5d9b129,389580759b..0000000000
--- a/users/manna-harbour_miryoku/manna-harbour_miryoku.c
+++ b/users/manna-harbour_miryoku/manna-harbour_miryoku.c
@@@ -17,14 -17,14 +17,22 @@@ MIRYOKU_LAYER_LIS
  #undef MIRYOKU_X
  };
  
++<<<<<<< HEAD
 +void u_td_fn_boot(qk_tap_dance_state_t *state, void *user_data) { \
++=======
+ void u_td_fn_boot(tap_dance_state_t *state, void *user_data) {
++>>>>>>> origin/master
    if (state->count == 2) {
      reset_keyboard();
    }
  }
  
  #define MIRYOKU_X(LAYER, STRING) \
++<<<<<<< HEAD
 +void u_td_fn_U_##LAYER(qk_tap_dance_state_t *state, void *user_data) { \
++=======
+ void u_td_fn_U_##LAYER(tap_dance_state_t *state, void *user_data) { \
++>>>>>>> origin/master
    if (state->count == 2) { \
      default_layer_set((layer_state_t)1 << U_##LAYER); \
    } \
@@@ -32,7 -32,7 +40,11 @@@
  MIRYOKU_LAYER_LIST
  #undef MIRYOKU_X
  
++<<<<<<< HEAD
 +qk_tap_dance_action_t tap_dance_actions[] = {
++=======
+ tap_dance_action_t tap_dance_actions[] = {
++>>>>>>> origin/master
      [U_TD_BOOT] = ACTION_TAP_DANCE_FN(u_td_fn_boot),
  #define MIRYOKU_X(LAYER, STRING) [U_TD_U_##LAYER] = ACTION_TAP_DANCE_FN(u_td_fn_U_##LAYER),
  MIRYOKU_LAYER_LIST
(END)
[... 182 lines cut ...]
 create mode 100644 .github/workflows/auto_approve.yml
 create mode 100644 .github/workflows/auto_tag.yml
 create mode 100644 .github/workflows/cli.yml
 create mode 100644 .github/workflows/develop_update.yml
 create mode 100644 .github/workflows/docs.yml
 create mode 100644 .github/workflows/feature_branch_update.yml
 create mode 100644 .github/workflows/format.yml
 create mode 100644 .github/workflows/format_push.yml
 create mode 100644 .github/workflows/labeler.yml
 create mode 100644 .github/workflows/lint.yml
 create mode 100644 .github/workflows/stale.yml
 create mode 100644 .github/workflows/unit_test.yml
 create mode 100644 readme.md

stratos@iocaine 
…/qmk_firmware [ miryoku-merge-master] [🐍 v3.10.10] [⍱ v2.3.4] 
➜  git fetch origin

stratos@iocaine 
…/qmk_firmware [ miryoku-merge-master] [🐍 v3.10.10] [⍱ v2.3.4] [⏱ 2s ] 
➜  git merge origin/master
Auto-merging users/manna-harbour_miryoku/manna-harbour_miryoku.c
CONFLICT (content): Merge conflict in users/manna-harbour_miryoku/manna-harbour_miryoku.c
Automatic merge failed; fix conflicts and then commit the result.
```

manna-harbour, 2023-04-07:
The miryoku branch needs rebasing, but for local builds QMK master should be up to date at the moment, so you can just build from QMK master directly.

manna-harbour, 2023-09-13:
The miryoku branch has been rebased and now merges cleanly with QMK master. Here's a sample build run: https://github.com/manna-harbour/miryoku_qmk/actions/runs/6171714664.

closed 2023-09-13 by manna-harbour