manna-harbour/miryoku #363: Error: Compiling: quantum/keymap_introspection.c

10bn, 2024-11-28:
I tried using Miryoku with the latest QMK branch to enable support for the “Azoteq Touchpad,” but I encountered an error during the build process.

Following the instructions from [this discussion](https://github.com/manna-harbour/miryoku/discussions/287), I copied the Miryoku userspace and the 3w6 keyboard folder from the Miryoku_QMK branch to the latest QMK branch.

```
  ~/code/keyboards/user_keymaps_still/qmk_firmware   playground !11 ?7                                                                7s  02:41:04 PM
❯ make 3w6/rev2:manna-harbour_miryoku:flash QMK_USER=users/manna-harbour_miryoku

QMK Firmware 0.27.0
☒ 3w6/rev2: LAYOUT: Layout macro should not be defined within ".h" files.
Making 3w6/rev2 with keymap manna-harbour_miryoku and target flash

☒ 3w6/rev2: LAYOUT: Layout macro should not be defined within ".h" files.
avr-gcc (Homebrew AVR GCC 8.5.0_2) 8.5.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   25650       0   25650    6432 3w6_rev2_manna-harbour_miryoku.hex

Generating: .build/obj_3w6_rev2_manna-harbour_miryoku/src/info_config.h                             [WARNINGS]
 | 
 | ☒ 3w6/rev2: LAYOUT: Layout macro should not be defined within ".h" files.
 | 
Generating: .build/obj_3w6_rev2_manna-harbour_miryoku/src/default_keyboard.h                        [WARNINGS]
 | 
 | ☒ 3w6/rev2: LAYOUT: Layout macro should not be defined within ".h" files.
 | 
Generating: .build/obj_3w6_rev2_manna-harbour_miryoku/src/default_keyboard.c                        [WARNINGS]
 | 
 | ☒ 3w6/rev2: LAYOUT: Layout macro should not be defined within ".h" files.
 | 
Compiling: quantum/keymap_introspection.c                                                          quantum/keymap_introspection.c: In function 'keymap_layer_count_raw':
quantum/keymap_introspection.c:18:49: error: 'keymaps' undeclared (first use in this function)
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:21:12: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
     return NUM_KEYMAP_LAYERS_RAW;
            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:18:49: note: each undeclared identifier is reported only once for each function it appears in
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:21:12: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
     return NUM_KEYMAP_LAYERS_RAW;
            ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: At top level:
quantum/keymap_introspection.c:18:49: error: 'keymaps' undeclared here (not in a function)
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                                                 ^~~~~~~
quantum/keymap_introspection.c:31:16: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT");
                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c:31:41: error: 'MAX_LAYER' undeclared here (not in a function)
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT");
                                         ^~~~~~~~~
quantum/keymap_introspection.c:18:31: error: expression in static assertion is not an integer
 #define NUM_KEYMAP_LAYERS_RAW ((uint8_t)(sizeof(keymaps) / ((MATRIX_ROWS) * (MATRIX_COLS) * sizeof(uint16_t))))
                               ^
quantum/keymap_introspection.c:31:16: note: in expansion of macro 'NUM_KEYMAP_LAYERS_RAW'
 _Static_assert(NUM_KEYMAP_LAYERS_RAW <= MAX_LAYER, "Number of keymap layers exceeds maximum set by LAYER_STATE_(8|16|32)BIT");
                ^~~~~~~~~~~~~~~~~~~~~
quantum/keymap_introspection.c: In function 'keycode_at_keymap_location_raw':
quantum/keymap_introspection.c:36:16: error: implicit declaration of function 'pgm_read_word' [-Werror=implicit-function-declaration]
         return pgm_read_word(&keymaps[layer_num][row][column]);
                ^~~~~~~~~~~~~
quantum/keymap_introspection.c:38:12: error: 'KC_TRNS' undeclared (first use in this function)
     return KC_TRNS;
            ^~~~~~~
quantum/keymap_introspection.c: In function 'keymap_layer_count_raw':
quantum/keymap_introspection.c:22:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
quantum/keymap_introspection.c: In function 'keycode_at_keymap_location_raw':
quantum/keymap_introspection.c:39:1: error: control reaches end of non-void function [-Werror=return-type]
 }
 ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
make[1]: *** [.build/obj_3w6_rev2_manna-harbour_miryoku/quantum/keymap_introspection.o] Error 1
make: *** [3w6/rev2:manna-harbour_miryoku:flash] Error 1
Make finished with errors

  ~/code/keyboards/user_keymaps_still/qmk_firmware   playground !11 ?7                                                                7s  02:42:19 PM
❯ 
```

winmillwill, 2024-11-30:
Looks like you need to apply the fix described here: https://docs.qmk.fm/ChangeLog/20240825#key-override-keymap-c-signature-change-24120

on the line described here:

```
In file included from quantum/keymap_introspection.c:9:
users/manna-harbour_miryoku/manna-harbour_miryoku.c:57:24: note: first 'sizeof' operand was declared here
 const key_override_t **key_overrides = (const key_override_t *[]){
                        ^~~~~~~~~~~~~
``` 

I just figured this out myself and all I can say is it made compilation complete without error, no idea yet whether the resulting firmware works as expected (and I'm new at this so no idea about much of anything really).

manna-harbour, 2024-12-13:
Also see https://github.com/manna-harbour/miryoku/discussions/361.